QuickLogic-Corp/symbiflow-arch-defs #29: "TypeError: list indices must be integers or slices, not str"  in timing importer

smunaut, 2020-06-24:
Trying to build aa65abdeb8754f60f425ca4ecfa03c176104e90f

```
[  0%] Generating sdf/BIDIR_COM_BC_180V.sdf
Traceback (most recent call last):
  File "/usr/lib64/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib64/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/tnt/projects/fpga/toolchain/symbiflow-arch-defs/quicklogic/utils/quicklogic-timings-importer/quicklogic_timings_importer/__main__.py", line 4, in <module>
    quicklogic_timings_importer.main()
  File "/home/tnt/projects/fpga/toolchain/symbiflow-arch-defs/quicklogic/utils/quicklogic-timings-importer/quicklogic_timings_importer/quicklogic_timings_importer.py", line 528, in main
    args.timescale,
  File "/home/tnt/projects/fpga/toolchain/symbiflow-arch-defs/quicklogic/utils/quicklogic-timings-importer/quicklogic_timings_importer/quicklogic_timings_importer.py", line 360, in export_sdf_from_lib_dict
    direction = obj['direction']
TypeError: list indices must be integers or slices, not str
make[2]: *** [quicklogic/devices/ql-pp3e-virt/CMakeFiles/file_quicklogic_devices_ql-pp3e-virt_sdf_BIDIR_COM_BC_180V.sdf.dir/build.make:63: quicklogic/devices/ql-pp3e-virt/sdf/BIDIR_COM_BC_180V.sdf] Error 1
make[1]: *** [CMakeFiles/Makefile2:10295: quicklogic/devices/ql-pp3e-virt/CMakeFiles/file_quicklogic_devices_ql-pp3e-virt_sdf_BIDIR_COM_BC_180V.sdf.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

smunaut, 2020-06-24:
I had to update the ql-pp3e submodule to make it build.
Fix is 2e1174aea2f52717570ef7522ac5483fe06363bd of that module

mkurc-ant, 2021-07-12:
Is this issue still relevant? In the recent effort for merging PP3 (EOS-S3) support into the SymbiFlow master branch care is taken to ensure that all submodules point to correct commits. Everything is verified by CI which hasn't reported an error similar to this so far.

smunaut, 2021-07-12:
Probably not.