QuickLogic-Corp/symbiflow-arch-defs #96: qlfasm file has hardcoded python path

kkumar23, 2020-07-27:
make install copies the qlfasm file to <install_path>/bin/python
The qlfasm file will retain the python3 path of the compilation path. It would result in error when user runs.

/mnt/c/Users/saxe/Documents/ActiveProjects/symbiflow/install/bin/write_bitstream: /mnt/c/Users/saxe/Documents/ActiveProjects/symbiflow/install/bin/python/qlfasm: /home/kkumar/symbiflow-arch-defs/build/env/conda/bin/python3: bad interpreter: No such file or directory'

acomodi, 2020-08-17:
If I am not mistaken, qlfasm can be installed in conda (the quicklogic-fasm project has a setup.py script) I believe that the copy of the qlfasm file is not required if it gets installed using conda.

acomodi, 2020-08-19:
Fixed with https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/124. Closing

closed 2020-08-19 by acomodi