QuickLogic-Corp/symbiflow-arch-defs #97: all_ql_tests target shows errors

kkumar23, 2020-07-27:
The target all_ql_tests shows up errors while running for various designs. Needs to be fixed.

acomodi, 2020-08-11:
After fixing the some of the issues with CI, I ran the `all_ql_tests` target using ninja and got the following failed tests, which can be divided in four sub-categories:

1. IO placement script failing to find a net:
```
FAILED: quicklogic/pp3/tests/consts/consts-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_io.place
FAILED: quicklogic/pp3/tests/lut/lut2-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_io.place
FAILED: quicklogic/pp3/tests/lut/lut4-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_io.place
FAILED: quicklogic/pp3/tests/lut/lut3-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_io.place
FAILED: quicklogic/pp3/tests/lut/lut1-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_io.place
FAILED: quicklogic/pp3/tests/quicklogic_testsuite/design3/design3-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_io.place
FAILED: quicklogic/pp3/tests/quicklogic_testsuite/design5/design5-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_io.place
```

The error returned looks like the following:
```
Traceback (most recent call last):
  File "/data/quicklogic/quicklogic-symbiflow/quicklogic/common/utils/create_ioplace.py", line 151, in <module>
    main()
  File "/data/quicklogic/quicklogic-symbiflow/quicklogic/common/utils/create_ioplace.py", line 115, in main
    pcf_constraint.net
  File "/data/quicklogic/quicklogic-symbiflow/utils/vpr_io_place.py", line 105, in get_top_level_block_instance_for_net
    block_name = self.net_to_block[net_name]
KeyError: 'I(0)'
ninja: build stopped: subcommand failed.
```

I believe this might be a synthesis issue. The problem here is that the `I(0)` input is unconnected in the .eblif, hence it will not be packed in the top.net, with the result of not being able to find the corresponding block.
This might be due to the yosys rebase or something yosys-related. Investigating

2. Resource utilization exceeds device limits
```
FAILED: quicklogic/pp3/tests/sdiomux_counter/sdiomux-counter-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/sdiomux_counter/sdiomux-counter-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/soc_clocks/soc_clocks-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/soc_clocks/soc_clocks-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/counter/counter-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/counter/counter-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/counter/counter-ql-quickfeather/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/counter/counter-ql-quickfeather/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/ram/ram-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/ram/ram-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/soc_clocks/soc_clocks-ql-quickfeather/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/soc_clocks/soc_clocks-ql-quickfeather/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/quicklogic_testsuite/iir/iir-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/quicklogic_testsuite/iir/iir-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/quicklogic_testsuite/design4/design4-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/quicklogic_testsuite/design4/design4-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/quicklogic_testsuite/design7/design7-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/quicklogic_testsuite/design7/design7-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/quicklogic_testsuite/cavlc_top/cavlc-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/quicklogic_testsuite/cavlc_top/cavlc-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/quicklogic_testsuite/sha256_fullfit/sha256_fullfit-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/quicklogic_testsuite/sha256_fullfit/sha256_fullfit-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
```

3. No routing convergence:
```
FAILED: quicklogic/pp3/tests/quicklogic_testsuite/design2/design2-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.route
```
This is related to https://github.com/QuickLogic-Corp/symbiflow-arch-defs/issues/110

4. Ram modeling:
```
FAILED: quicklogic/pp3/devices/ql-pp3e-virt/ram.model.xml quicklogic/pp3/devices/ql-pp3e-virt/ram.pb_type.xml quicklogic/pp3/devices/ql-pp3e-virt/ram_sim.v quicklogic/pp3/devices/ql-pp3e-virt/ram_map.v
```

I have a local fix for this. I found that there is a copy happening in the CMake command that is static, and is relative to the ql-eos device only and does not work for the pp3: https://github.com/QuickLogic-Corp/symbiflow-arch-defs/blob/da040dd93fca9d6928b77e30987e6b6b88b69e14/quicklogic/common/cmake/quicklogic_device.cmake#L166

I am not sure why that copy command is needed at all actually. Shouldn't the ram.pb_type.xml be generated with the command above the copy?

acomodi, 2020-08-11:
Update:

The error 4) is solved here https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/113.

@kkumar23 Do you happen to know what is the correct timing file for the ram of the ql-PP3 device? At the moment I am have set RAM_COM_BC_120V

Also, do you recall why the following line was needed? https://github.com/QuickLogic-Corp/symbiflow-arch-defs/blob/da040dd93fca9d6928b77e30987e6b6b88b69e14/quicklogic/common/cmake/quicklogic_device.cmake#L166

kkumar23, 2020-08-12:
Travis CI is in red for this commit.

acomodi, 2020-08-12:
> Travis CI is in red for this commit.

@kkumar23 Fixed with https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/117

acomodi, 2020-08-13:
If using https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/121 the remaining failing tests are all relative to the packer. The failures during packing can be divided in two categories:

1. Issue with constant nets that do not have a driver. This happens prior to packing at it occurs when reading the netlist (could be a synthesis issue or a circuit cleaning issue made in VPR, unsure at the moment).
    failure example:
```
vpr/src/base/read_netlist.cpp:1079 mark_constant_generators_rec: Assertion 'atom_ctx.nlist.pin_is_constant(driver_pin_id)' failed.   
``` 

This happens for the following targets:

```
FAILED: quicklogic/pp3/tests/counter/counter-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/counter/counter-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/soc_clocks/soc_clocks-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/soc_clocks/soc_clocks-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/sdiomux_counter/sdiomux-counter-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/sdiomux_counter/sdiomux-counter-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/counter/counter-ql-quickfeather/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/counter/counter-ql-quickfeather/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/soc_clocks/soc_clocks-ql-quickfeather/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/soc_clocks/soc_clocks-ql-quickfeather/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/ram/ram-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/ram/ram-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log

```

2. Not enough resources. 
With the last two commits of #121 (https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/121/commits/17c411c25190a6bd20d2933850e3caddb16978ae and https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/121/commits/da794e8af890f03576829f9dc83c7acff0f17a15), the mux8 are now fragmented in two mux4 and an F_FRAG, as previously, by having them implemented in a C_FRAG cause the unroutability due to extreme congestion for the `quicklogic_testsuite/design2`.
There is also the addition of a packer parameter that adds a threshold when packing nets into a PB-LOGIC.
Both the two commits slightly increase the number of blocks generated by the packer, but the following targets were already failing prior to these updates:

```
FAILED: quicklogic/pp3/tests/quicklogic_testsuite/iir/iir-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/quicklogic_testsuite/iir/iir-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/quicklogic_testsuite/cavlc_top/cavlc-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/quicklogic_testsuite/cavlc_top/cavlc-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/quicklogic_testsuite/design4/design4-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/quicklogic_testsuite/design4/design4-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/quicklogic_testsuite/sha256_fullfit/sha256_fullfit-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/quicklogic_testsuite/sha256_fullfit/sha256_fullfit-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
FAILED: quicklogic/pp3/tests/quicklogic_testsuite/design7/design7-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net quicklogic/pp3/tests/quicklogic_testsuite/design7/design7-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/pack.log
```

acomodi, 2020-08-13:
Regarding 1) from the above comment though, the issue might be later in the flow, and not related to the synthesized net, as it happens after generating the packed netlist:

```
Complex block 0: 'u_qlal4s3b_cell_macro' (PB-ASSP) .
Complex block 1: 'cnt_LUT2_I0_1.t_frag' (PB-LOGIC) ....
Complex block 2: 'cnt_LUT3_I0_10.t_frag' (PB-LOGIC) ...
Complex block 3: 'cnt_LUT3_I0_14.t_frag' (PB-LOGIC) ...
Complex block 4: 'cnt_LUT3_I0_18.t_frag' (PB-LOGIC) ...
Complex block 5: 'cnt_LUT3_I0_2.t_frag' (PB-LOGIC) ...
Complex block 6: 'cnt_LUT3_I0_6.t_frag' (PB-LOGIC) ...
Complex block 7: 'cnt_LUT2_I0_2.t_frag' (PB-LOGIC) ...
Complex block 8: 'cnt_LUT2_I0_3.t_frag' (PB-LOGIC) ...
Complex block 9: '$iopadmap$top.led' (PB-BIDIR) .
Complex block 10: '$iopadmap$top.led_1' (PB-BIDIR) .
Complex block 11: '$iopadmap$top.led_2' (PB-BIDIR) .
Complex block 12: '$iopadmap$top.led_3' (PB-BIDIR) .
Complex block 13: 'cnt_LUT2_I0_4.t_frag' (PB-LOGIC) ...
Complex block 14: 'cnt_LUT2_I0_5.t_frag' (PB-LOGIC) ...
Complex block 15: 'cnt_dff_Q_1' (PB-LOGIC) ...
Complex block 16: 'cnt_LUT2_I0_6.t_frag' (PB-LOGIC) ...
Complex block 17: 'cnt_dff_Q_5' (PB-LOGIC) ...
Complex block 18: 'cnt_LUT2_I0_8.t_frag' (PB-LOGIC) ...
Complex block 19: 'cnt_dff_Q_7' (PB-LOGIC) ...
Complex block 20: 'cnt_LUT2_I0_10.t_frag' (PB-LOGIC) ...
Complex block 21: 'cnt_dff_Q_11' (PB-LOGIC) ...
Complex block 22: 'cnt_dff_Q_12' (PB-LOGIC) .
Complex block 23: 'cnt_dff_Q_15' (PB-LOGIC) .
Complex block 24: 'cnt_dff_Q_16' (PB-LOGIC) .
Complex block 25: 'cnt_dff_Q_19' (PB-LOGIC) .
Complex block 26: 'cnt_dff_Q_20' (PB-LOGIC) .
Complex block 27: 'cnt_dff_Q_21' (PB-LOGIC) .
Complex block 28: 'cnt_dff_Q_23' (PB-LOGIC) .
Complex block 29: 'u_gclkbuff_clock' (PB-GMUX) .
Complex block 30: '$false' (PB-SYN_GND) .
Complex block 31: '$true' (PB-SYN_VCC) .
        EMPTY: # blocks: 0, average # input + clock pins used: 0, average # output pins used: 0
        PB-ASSP: # blocks: 1, average # input + clock pins used: 0, average # output pins used: 1
        PB-SYN_VCC: # blocks: 1, average # input + clock pins used: 0, average # output pins used: 1
        PB-MULT: # blocks: 0, average # input + clock pins used: 0, average # output pins used: 0
        PB-CLOCK: # blocks: 0, average # input + clock pins used: 0, average # output pins used: 0
        PB-RAM: # blocks: 0, average # input + clock pins used: 0, average # output pins used: 0
        PB-LOGIC: # blocks: 24, average # input + clock pins used: 15.3333, average # output pins used: 2.45833
        PB-SYN_GND: # blocks: 1, average # input + clock pins used: 0, average # output pins used: 1
        PB-BIDIR: # blocks: 4, average # input + clock pins used: 3, average # output pins used: 0
        PB-SDIOMUX: # blocks: 0, average # input + clock pins used: 0, average # output pins used: 0
        PB-GMUX: # blocks: 1, average # input + clock pins used: 2, average # output pins used: 1
Absorbed logical nets 4 out of 67 nets, 63 nets not absorbed.
FPGA sized to 39 x 35 (ql-eos-s3)
Device Utilization: 0.02 (target 1.00)
        Block Utilization: 1.00 Type: PB-ASSP
        Block Utilization: 1.00 Type: PB-SYN_VCC
        Block Utilization: 0.03 Type: PB-LOGIC
        Block Utilization: 1.00 Type: PB-SYN_GND
        Block Utilization: 0.12 Type: PB-BIDIR
        Block Utilization: 0.20 Type: PB-GMUX

Netlist conversion complete.

# Packing took 0.04 seconds (max_rss 40.4 MiB, delta_rss +0.6 MiB)
# Load Packing
Begin loading packed FPGA netlist file.
Netlist generated from file 'top.net'.
/home/kkumar/anaconda3/conda-bld/vtr_1596098288083/work/vpr/src/base/read_netlist.cpp:1079 mark_constant_generators_rec: Assertion 'atom_ctx.nlist.pin_is_constant(driver_pin_id)' failed.
Aborted (core dumped)
```

acomodi, 2020-08-19:
Closing this issue as it is superseded by https://github.com/QuickLogic-Corp/symbiflow-arch-defs/issues/132

closed 2020-08-19 by acomodi