QuickLogic-Corp/symbiflow-arch-defs #109: Significant Degradation in Timing with Utilization improvements

tpagarani, 2020-08-09:
Recently there have been PR merged to improve the Utilization.

https://github.com/QuickLogic-Corp/yosys/pull/10

There is significant degradation in Delay parameters like Critical path, TNS. 
I have attached the report with and without recent improvements in mux and pack. 

[07-28-2020_design_run_results.xlsx](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/5047672/07-28-2020_design_run_results.xlsx)
[08-02-2020_design_run_results.xlsx](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/5047673/08-02-2020_design_run_results.xlsx)

mkurc-ant, 2020-08-24:
This issue can be mitigated by enabling LUT4 inference in ABC and increasing their cost via `abc -luts 1,2,2,4:8`. The drawback is increased pos-synthesis design sizes.

Here are the results: [results.xlsx](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/5118205/results.xlsx)

mkurc-ant, 2020-08-26:
Should have been fixed by https://github.com/QuickLogic-Corp/yosys/pull/14

tpagarani, 2020-09-29:
This issue was fixed

closed 2020-09-29 by tpagarani