QuickLogic-Corp/symbiflow-arch-defs #110: Design failing in router

tpagarani, 2020-08-09:
After implementation of Mux and pack optimization in Yosys, following design is passing through packer however it fails in router

https://github.com/QuickLogic-Corp/symbiflow-arch-defs/tree/quicklogic-upstream-rebase/quicklogic/pp3/tests/quicklogic_testsuite/design2

"Circuit is unroutable with a channel width factor of 100. VPR failed to implement circuit"

acomodi, 2020-08-13:
This should be fixed with the last two commits of https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/121 (17c411c and da794e8). More details on this issue comment: https://github.com/QuickLogic-Corp/symbiflow-arch-defs/issues/97#issuecomment-673560981

acomodi, 2020-08-19:
I have double checked and re-run the design2 test and, with the current quicklogic-upstream-rebase status (https://github.com/QuickLogic-Corp/symbiflow-arch-defs/commit/5a6f743328c735c33f4aff760ffa524febbb6559) the tests correctly routes

closed 2020-08-19 by acomodi