QuickLogic-Corp/symbiflow-arch-defs #160: Unable to route the design [tests/quicklogic_testsuite/wbqspiflash]

kkumar23, 2020-09-11:
The design tests/quicklogic_testsuite/wbqspiflash, fails while routing:
Circuit is unroutable with a channel width factor of 100.
VPR failed to implement circuit
The entire flow of VPR took 2349.41 seconds (max_rss 631.6 MiB)
+ rm vpr.output.IkH

kgugala, 2020-09-16:
I ran the test targeting Quickfeather using `3393d6b32cbb3ecc779b7c0a953fd6963987e1a2` arch-defs commit. With VPR from conda channel and quicklogic-yosys commit `8a51c9cbb843fd1cf75727fa08f8a0314e97c98b`.

I was able to generate bitstream. Will also check this for chandalar.

@kkumar23 can you list the exact version of the tools you're using?

kgugala, 2020-09-16:
chandalar test also generates bitstream with the above tools revisions

kkumar23, 2020-09-16:
@kgugala It is routing with latest version. However the release version does not route. We can close this,

closed 2020-09-16 by kkumar23