QuickLogic-Corp/symbiflow-arch-defs #171: make install target has issues

kkumar23, 2020-10-15:
make install DESTDIR=<install location>
Shows the below error:
EOFError: Ran out of input
quicklogic/pp3/CMakeFiles/file_quicklogic_pp3_chandalar_clkmap.csv.dir/build.make:81: recipe for target 'quicklogic/pp3/chandalar_clkmap.csv' failed
quicklogic/pp3/CMakeFiles/file_quicklogic_pp3_quickfeather_clkmap.csv.dir/build.make:81: recipe for target 'quicklogic/pp3/quickfeather_clkmap.csv' failed

acomodi, 2020-10-15:
Hi @kkumar23,
Running 
```
DESTDIR=<install_dir> ninja install
```

Worked correctly for me on https://github.com/QuickLogic-Corp/symbiflow-arch-defs/commit/a7c45b18e26415fe097e28a651263c62bdeb75e1.

Can you please provide a version of SymbiFlow arch defs that is generating the error, and the exact steps to reproduce?

acomodi, 2020-10-15:
I could reproduce the issue using `make` instead of `ninja` with a number of jobs > 1.

I recall having the same issue when trying to get CI green.

The fact is that, having multiple jobs build, generates dependency issues with the clkmap generation as the script is being run even though the database creation has not yet completed the `pickle` file writing process, ending up in the following error:
```
Traceback (most recent call last):
  File "/data/quicklogic/quicklogic-symbiflow/quicklogic/common/utils/create_clkmap_csv.py", line 107, in <module>
    main()
  File "/data/quicklogic/quicklogic-symbiflow/quicklogic/common/utils/create_clkmap_csv.py", line 91, in main
    db = pickle.load(args.db)
_pickle.UnpicklingError: pickle data was truncated
quicklogic/pp3/CMakeFiles/file_quicklogic_pp3_quickfeather_clkmap.csv.dir/build.make:62: recipe for target 'quicklogic/pp3/quickfeather_clkmap.csv' failed
make[2]: *** [quicklogic/pp3/quickfeather_clkmap.csv] Error 1
make[2]: *** Deleting file 'quicklogic/pp3/quickfeather_clkmap.csv'
CMakeFiles/Makefile2:281893: recipe for target 'quicklogic/pp3/CMakeFiles/file_quicklogic_pp3_quickfeather_clkmap.csv.dir/all' failed
make[1]: *** [quicklogic/pp3/CMakeFiles/file_quicklogic_pp3_quickfeather_clkmap.csv.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File "/data/quicklogic/quicklogic-symbiflow/quicklogic/common/utils/create_clkmap_csv.py", line 107, in <module>
    main()
  File "/data/quicklogic/quicklogic-symbiflow/quicklogic/common/utils/create_clkmap_csv.py", line 91, in main
    db = pickle.load(args.db)
EOFError: Ran out of input
Traceback (most recent call last):
  File "/data/quicklogic/quicklogic-symbiflow/quicklogic/common/utils/create_clkmap_csv.py", line 107, in <module>
    main()
  File "/data/quicklogic/quicklogic-symbiflow/quicklogic/common/utils/create_clkmap_csv.py", line 91, in main
    db = pickle.load(args.db)
EOFError: Ran out of input
Traceback (most recent call last):
  File "/data/quicklogic/quicklogic-symbiflow/quicklogic/common/utils/create_clkmap_csv.py", line 107, in <module>
    main()
  File "/data/quicklogic/quicklogic-symbiflow/quicklogic/common/utils/create_clkmap_csv.py", line 91, in main
    db = pickle.load(args.db)
EOFError: Ran out of input
```

There are two ways to overcome this issue before a more stable fix is found:
1. Running single-job build
2. Using ninja instead of make: `make env CMAKE_FLAGS="-GNinja"`

kkumar23, 2020-10-15:
@acomodi : Here are the details:
Branch : quicklogic-upstream-rebase
https://github.com/QuickLogic-Corp/symbiflow-arch-defs/commit/a7c45b18e26415fe097e28a651263c62bdeb75e1
make env;make all_conda
make DESTDIR="some location" install

Ubuntu 18.04 LTS (Windows VM)

kkumar23, 2020-10-15:
@acomodi : Single job may take more time. As of now i can proceed. Any plan to fix this?
ninja i am using for Travis, i can use for local systems.

acomodi, 2020-10-15:
@kkumar23 I am looking into it. I think dependencies for the clkmap and pinmap files should be on the targets rather then the generated files.

kkumar23, 2020-10-15:
@acomodi : With the new vpr, i get the below error when run through the make install toolchain scripts:

Missing expected argument for --fix_pins

usage: vpr architecture circuit [--pack] [--place] [--route] [--analysis]
       [--disp {on, off}] [--save_graphics {on, off}]
       [--graphics_commands GRAPHICS_COMMANDS] [-h] [--version]
       [--device DEVICE_NAME] [-j NUM_WORKERS] [--timing_analysis {on, off}]
       [--disable_errors DISABLE_ERRORS] [--suppress_warnings SUPPRESS_WARNINGS]
       [--route_chan_width CHANNEL_WIDTH] [OTHER_OPTIONS ...]

steps to re-produce:
cd and example say counter_16bit:
and run:
ql_symbiflow -compile -d ql-eos-s3 -P pd64 -v counter_16bit.v -t top -p chandalar.pcf

acomodi, 2020-10-15:
@kkumar23 This commit solves the issue: https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/172/commits/19395df7e9f83b371d2ebba72d3905e193e6a5cc

You can test it locally by manually changing the symbiflow_place toolchain wrapper and apply the same fix

kkumar23, 2020-10-15:
@acomodi : Thanks.

closed 2020-10-28 by kkumar23