QuickLogic-Corp/symbiflow-arch-defs #234: Multi corner timing support

lnsharma, 2021-05-03:
When multiple vpr_arch.xml files are specified, corresponding to a specific timing corner delay data in it, then symbiflow should run the flow considering CORNER as an input and based on this input, it should run the flow by picking the corresponding vpr_arch.xml.

lnsharma, 2021-05-04:
Following are the different items to be supported for this issue:

1. Write a separate vpr_Arch.xml file (using the specified corner delay data) in OpenFPGA-QLF_K4N8 repo for different corners for qlf_k4n8 device. Generate bitstream database and rr_graph related files.
2. Update bitstream database and rr_graph related files for each corner in qlfpga-symbiflow-plugins repo
3. Introduce a new variable CORNER and update cmake structure in Symbiflow to pick vpr_arch.xml file accordingly.
4. Update ql_symbiflow wrapper to take CORNER as an option and run the follow for that corner.
5. Write testcases to verify if the flow works fine using wrapper or test targets

kkumar23, 2021-05-18:
@acomodi : I believe it is under progress. Please add relevant testcases along with the PR.

acomodi, 2021-05-18:
@kkumar23 Actually this was already merged and the testcases already do make use of one of the corners. The only missing thing is actually the correct corner data (I have duplicated the numbers at the moment, and the fast and slow corners share the same delays)

closed 2021-05-18 by acomodi