QuickLogic-Corp/symbiflow-arch-defs #235: The IO/clock constraint does not get honored at certain corner cases

kkumar23, 2021-05-03:
For a design where IO is followed by a Flipflop, the FF is pulled into the IO and registered IO is inferred. 
In this case if user wants to constraint the IO clk to some specific clock then the constraint is not honored

mkurc-ant, 2021-05-04:
@kkumar23 Could you please provide more details like eg. in which of the test designs you see this issue?

I couldn't have reproduced it so far. I took the `io_reg` test design and constrainted `clk` to `io.clk[3]`. Then, I verified the repacked netlist (.net) and confirmed that indeed the net `clk` enters the `io` block through `io.clk[3]`.

I've also verified the output FASM file. It contains the only line responsible for the clock input mux:
```
fpga_top.grid_io_bottom_bottom_5__0_.logical_tile_io_mode_io__7.mem_iopad_0_clk_0.mem_out[1]
```
which corresponds to the choice of `clk[3]` input.

I did the test for the current master branch (`90fd7572`).

kkumar23, 2021-05-04:
@mkurc-ant : I reinstalled the package and verified. I see the changes in the fasm file . It was for the io_reg design.

closed 2021-05-04 by kkumar23

kkumar23, 2021-05-04:
@mkurc-ant : can you add the make support for the json. We are using toolchain wrapper as of now.

mkurc-ant, 2021-05-04:
@kkumar23 Of course, I will be looking into that.