QuickLogic-Corp/symbiflow-arch-defs #237: Specify user-defined pin names in SDC

lnsharma, 2021-05-03:
In eFPGA flow, user should be able to specify user-defined pin-names (as given in csv file) in SDC commands. Currently, SDC commands only accepts device interface port names but not user-defined pin names which are mapped to interface ports.

acomodi, 2021-05-06:
@lnsharma To be clear, the intention is to constrain the clocks based on the pin names rather than the net names?

E.g.:
- clock net: clk
- pin associated to the clk: A3
- SDC command: `create_clock A3 -period <period>`

At the moment, VPR applies clock constraints on nets, not on pins.

acomodi, 2021-05-07:
Ok, thanks for the clarification

mkurc-ant, 2021-05-12:
@lnsharma So effectively you want a user to be able to provide chip pin names in SDC like in one of the examples from the pin mapping tutorial:
```
set_output_delay 10 -max -clock SYS_CLK_2 [get_ports F1]
```

Do you also want to allow users to provide net names instead pin names (as it is currently supported) too or you just want to support only pin names ?

closed 2021-06-11 by kkumar23