QuickLogic-Corp/symbiflow-arch-defs #238: Specify clock constraints in pcf file

lnsharma, 2021-05-03:
Currently clock constraints are processed through a json file. PCF file should be used even for specifying clock constraints.

mkurc-ant, 2021-05-11:
@lnsharma Could you elaborate more on how do you want to refer to clock inputs in PCF? I haven't seen any mention of any dedicatec clock input pins in any of the pinmap files.

I presume that global clock inputs are implicit because we don't route clocks in VPR. But we need to known how they are connected to the `clb` and `io` tile pins so that we can map PCF constraints to repacker constraints.

mkurc-ant, 2021-05-11:
@lnsharma So essentially you want to constrain the clock in PCF mostly the same way as it is now done via JSON file? So that a user specifies `set_clock <net name> <clock name>`?

Would it be ok if we name the clocks `clk0`, `clk1`, `clk2` and `clk3` (uppercase would also be ok)? This way we will be able to map them 1-to-1 to repacker constraints.

lnsharma, 2021-05-11:
> @lnsharma So essentially you want to constrain the clock in PCF mostly the same way as it is now done via JSON file? So that a user specifies `set_clock <net name> <clock name>`?
>
> Would it be ok if [...]

Sure, you can rename that. I just need to ensure that generated lib file should have the similar clock names.

kkumar23, 2021-05-18:
verified the clk constraints in the pcf.

closed 2021-05-18 by kkumar23