QuickLogic-Corp/symbiflow-arch-defs #244: Testcases failing at router

kkumar23, 2021-05-05:
Below testcases fail for qlf_k4n8 at router:
#add_subdirectory(multi_enc_decx2x4)
#add_subdirectory(unsigned_mult_60)

mkurc-ant, 2021-05-12:
The design `multi_enc_decx2x4` fails on routing due to congestion. There are two possible things that could solve the problem:
 - Fixing https://github.com/QuickLogic-Corp/symbiflow-arch-defs/issues/242
 - Re-enabling buffer LUT absorption in VPR followed by fixing all the issues caused by it in the repacker

I couldn't find the `unsigned_mult_60` test desgin on current master branch.

kkumar23, 2021-05-12:
Here are more designs which fail at router :
designs/unsigned_mult_80 
designs/e_sdio_host_controller 
designs/cavlc_top 

@tpagarani : can comment on the approach to take

mkurc-ant, 2021-05-27:
@kkumar23 I added the following tests to my WIP branch where I'm integrating enabled CLB input equivalence:
- multi_enc_decx2x4
- unsigned_mult_80
- e_sdio_host_controller
- cavlc_top
- sdio_client_top

All of them have passed routing.

Although the test under `features/clk_verify` fails. The actual failing check looks for the following regex in `route.log`:
```
egrep -q "Fmax: [3-4][0-9]." ${ROUTE_LOG}
```
So it basically expect fmax to be in range from 30 to 49 MHz. But with the pin equivalence enabled I'm getting:
```
Final critical path delay (least slack): 11.661 ns, Fmax: 85.7559 MHz
```
which is higher. Shall I update the regex so that it looks for the fmax that I'm getting now?

kkumar23, 2021-05-27:
@mkurc-ant : Yes, keep the range +/- 20%

mkurc-ant, 2021-06-02:
Ok, with https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/309 merged all the test cases pass. In that PR I've updated the reference max frequency for `features/clk_verify`

closed 2021-06-08 by kkumar23