QuickLogic-Corp/symbiflow-arch-defs #246: LUT value for the buf implementation is incorrect

rakeshm75, 2021-05-05:
The LUT value for the buf implementation is incorrect (its order reversed). I have attached the repacked eblif file (top.repacked.net) here.

.subckt frac_lut4_arith in[0]=u_cnt3.count_dffr_Q_D[6] lut4_out=u_cnt3.count_dffr_Q_D[6]$buf0
.param LUT 0100000000000000
.param MODE 0
.subckt frac_lut4_arith in[0]=u_cnt3.count_dffr_Q_D[5] lut4_out=u_cnt3.count_dffr_Q_D[5]$buf0
.param LUT 0100000000000000
.param MODE 0
.subckt frac_lut4_arith in[2]=u_cnt3.count_dffr_Q_D[4] lut4_out=u_cnt3.count_dffr_Q_D[4]$buf0
.param LUT 0000111100000000
.param MODE 0
.subckt frac_lut4_arith in[1]=u_cnt3.count_dffr_Q_D[3] lut4_out=u_cnt3.count_dffr_Q_D[3]$buf0
.param LUT 0011000000000000
.param MODE 0
.subckt frac_lut4_arith in[1]=u_cnt3.count_dffr_Q_D[1] lut4_out=u_cnt3.count_dffr_Q_D[1]$buf0
.param LUT 0011000000000000

[counter_design.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/6426653/counter_design.zip)

mkurc-ant, 2021-05-05:
@rakeshm75 How have you discovered that it is incorrect? Via simulation done before FASM/bitstream generation of via incorrect bitstream? I'm asking to determine where the issue may be.

Are only buffer LUTs affected or do you see the problem in LUTs used for logic too ?

rakeshm75, 2021-05-05:
Yes, through FASM force bit simulation. Initially, I simulated 16bit counter design and it worked fine. That design had synchronous reset and enable. This is also a counter design (4 8bit counter working on different clocks) with just an asynchronous reset and it failed. And on debug, found the LUT value implemented for buf was wrong in this design. 
So I cross checked how did the first design (16bit counter) work fine and found that design has implementation of synchronous reset and enable (which is correct) instead of the buf implementation in the current design.

tpagarani, 2021-05-05:
@mkurc-ant I am suspecting that LUT table for buffers is not correct. We might have to look at the rotation table 

https://github.com/QuickLogic-Corp/symbiflow-arch-defs/blob/master/quicklogic/qlf_k4n8/utils/repacker/repack.py#L192
Since the input port mapping in case of our LUT4 is reversed of .names

mkurc-ant, 2021-05-05:
Right, let me look into the repacker code. I suspect that the issue is there.

mkurc-ant, 2021-05-12:
With the fix from https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/253 the issue should no longer emerge.

kkumar23, 2021-05-18:
@rakeshm75 : If verified please close this issue

rakeshm75, 2021-05-27:
This issue is fixed, so closing this issue.

closed 2021-05-27 by rakeshm75