QuickLogic-Corp/symbiflow-arch-defs #259: post layout simulation fail

kkumar23, 2021-05-12:
I ran simualtion for the below design on the file:
the post layout file <top module>_post_synthesis.v for the design 

 counter_16bit design. : PASS
 counter_4clk design : FAIL 

SDF file when used: It shows the below error:
** Error (suppressible): (vsim-SDF-12090) D:/Open_core/symbiflow_MAIN/Simulation/post_layout/top_post_synthesis.sdf(2628): Failed to find matching specify timing constraint for SETUP. 
# ** Error (suppressible): (vsim-SDF-12090) D:/Open_core/symbiflow_MAIN/Simulation/post_layout/top_post_synthesis.sdf(2630): Failed to find matching specify timing constraint for HOLD. 
# ** Error (suppressible): (vsim-SDF-12090) D:/Open_core/symbiflow_MAIN/Simulation/post_layout/top_post_synthesis.sdf(2631): Failed to find matching specify timing constraint for HOLD.

mkurc-ant, 2021-05-13:
@kkumar23 I've build post-layout files for the `counter_4clk` design. I cannot see anything suspicious at lines 2630 and 2631 in the SDF:
```
    (CELL
        (CELLTYPE "frac_lut4_arith")
        (INSTANCE frac_lut4_arith_cnt1_adder_lut4_in_5)
        (DELAY
            (ABSOLUTE
                (IOPATH in[0] lut4_out (330:330:330))
                (IOPATH in[1] lut4_out (330:330:330))
                (IOPATH in[2] lut4_out (330:330:330))
                (IOPATH in[3] lut4_out (330:330:330)) <- line 2630
                (IOPATH cin lut4_out (249:249:249)) <- line 2631
                (IOPATH in[0] cout (202:202:202))
                (IOPATH in[1] cout (202:202:202))
                (IOPATH in[2] cout (202:202:202))
                (IOPATH in[3] cout (202:202:202))
                (IOPATH cin cout (47:47:47))
            )
        )
    )
```

Could you please attach the files that you are using for the simulation? Also please tell what software you are using.

kkumar23, 2021-05-13:
Hi @mkurc-ant : I am using questasim tool. 
The design is failing functionally[even if i do not use sdf]
With sdf i get the errors: 
[questasim.log](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/6471513/questasim.log)
[top_post_synthesis.txt](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/6471517/top_post_synthesis.txt)

closed 2021-06-02 by kkumar23