QuickLogic-Corp/symbiflow-arch-defs #305: Shift register design which is not multiple of 8 register is not working

rakeshm75, 2021-05-27:
Tool reports error for shift register design which has registers not multiple of 8. Attached is a shift register design with 12 flops (12 registers). The tool reports the following error:

Error 1: 
Type: Packing
File: /home/runner/work/conda-eda/conda-eda/workdir/conda-env/conda-bld/vtr-optimized_1616063160078/work/vpr/src/pack/cluster.cpp
Line: 2337
Message: Can not find any logic block that can implement molecule.
	Pattern shiftchain shift_sh_dff_D

Net 'shift_out' is impossible to route within proposed clb cluster
Failed route at end, repack cluster trying detailed routing at each stage.
Net 'shift_out' is impossible to route within proposed clb cluster
Routing was impossible!
Net 'shift_out' is impossible to route within proposed clb cluster
Incr Slack updates 1 in 7.275e-06 sec
Full Max Req/Worst Slack updates 1 in 2.924e-06 sec
Incr Max Req/Worst Slack updates 0 in 0 sec
Incr Criticality updates 0 in 0 sec
Full Criticality updates 1 in 1.1445e-05 sec
# Packing took 0.02 seconds (max_rss 10.1 MiB, delta_rss +1.2 MiB)
The entire flow of VPR took 0.05 seconds (max_rss 10.1 MiB)

[shft_reg12.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/6554323/shft_reg12.zip)

mkurc-ant, 2021-06-02:
It looks like this is an issue with VPR chain molecule packing. The `sh_dff` cell has only one output port that is used for chaining hence VPR cannot route it out of a cluster in a mid-chain position. There is also no possibility to begin a chain in the middle of a physical 8-long chain.

To solve that probably a second virtual output port needs to be added to `sh_dff`. Then the repacker may take care of mapping it back to the physical flip-flop output.

mkurc-ant, 2021-06-02:
This PR adds a connection to the VPR architecture that allows VPR to utilize chains that are not a multiple of 8: https://github.com/QuickLogic-Corp/OpenFPGA-QLF_K4N8/pull/211

mkurc-ant, 2021-07-12:
@kkumar23 The change in architecture merged with the PR from the previous comment should have fixed the issue. From now on shift regs that are not a multiple of 8 should work correctly.

kkumar23, 2021-07-13:
@rakeshm75 : Can you please validate this and close?

mkurc-ant, 2021-07-13:
I've created a test with a shift register of length 12. Please take a look: https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/416

rakeshm75, 2021-07-16:
Issue resolved.

closed 2021-07-16 by rakeshm75