QuickLogic-Corp/symbiflow-arch-defs #322: Make install support for qlf_k6n10

kkumar23, 2021-06-02:
make install is failing for k6n10.
Also tool chain wrapper support is required for k6n10 till router.

kkumar23, 2021-06-03:
Toolchain wrappers i have upadated https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/324 please review.
For k6n10 make install, does not install scripts and techmaps folders.
fasm is still not supported for k6n10.

kkumar23, 2021-06-04:
@mkurc-ant : few more things to be added will be the package uploaded to have both k4n8 and k6n10 files. 
We need to provide build to customer today, so please consider this as priority.

mkurc-ant, 2021-06-04:
@kkumar23 Ok, you can merge #324. I'll check and fix CI so that data files for k6n10 are included in the package

closed 2021-06-11 by kkumar23