QuickLogic-Corp/symbiflow-arch-defs #460: Add bitstream generation support for PP3

acomodi, 2021-08-11:
After adding and initial support for the PP3 in the arch-defs flow (https://github.com/QuickLogic-Corp/symbiflow-arch-defs/issues/459), the next step is to get support for bitstream generation as well.

mkurc-ant, 2021-09-02:
The `quicklogic-fasm` utility has been generalized to support `EOS-S3`, `PP3` and `PP3E` in the future with https://github.com/QuickLogic-Corp/quicklogic-fasm/pull/14. For that the `QL732B-cdl.cmd` file has been checked in for EOS-S3: https://github.com/QuickLogic-Corp/EOS-S3/pull/10

The PR with SymbiFlow integration (CMake flow only) is open: https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/488

lpawelcz, 2021-09-07:
https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/488 has been merged. Working on bitstream generation support in:  https://github.com/QuickLogic-Corp/quicklogic-fasm/pull/15

anthonyle-ql, 2021-09-24:
I tested the latest Symbiflow with PP3 and there is no bit file generated.

Command: ql_symbiflow -compile -d ql-pp3 -P WD30 -v counter_8bit.v -t top -p wd30.pcf

Attached is all related files.
[test.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/7222134/test.zip)