QuickLogic-Corp/symbiflow-arch-defs #462: Add PP3 in the toolchain installation

acomodi, 2021-08-11:
As soon as the whole PP3 supported is thoroughly verified with fasm2bels and the various testsuites, the PP3 arch can be added to the installation package, and it would need to have a set of installation tests to verify that everything is working as expected.

Depends on:
#459 
#460 
#461

mkurc-ant, 2021-09-02:
There has been an issue discovered with the CI tests on `QuickLogic-Corp/symbiflow-arch-defs`. The scripts were not testing the installed toolchain properly hence some targets for `EOS-S3` were not working properly.

The ongoing work includes fixing the CI followed with adding `PP3` support to the toolchain wrappers.

mkurc-ant, 2021-09-08:
With the PR (https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/488) merged both the CI got fixed and PP3 support was added to the installed toolchain.