QuickLogic-Corp/symbiflow-arch-defs #534: Daily build: Counter_16bit fails VPR step

anthonyle-ql, 2021-09-21:
I am able to build the code and run the command below. I got errors and would like to report the error. Where is the right place to report this error? Attached is VPR log file.

[vpr_stdout.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/7200055/vpr_stdout.zip)

anthonyle-ql, 2021-09-22:
I change the command to: ql_symbiflow -compile -d ql-pp3 -t top -v counter_16bit.v -p counter_16bit.pcf
The tool is able to go further than the VPR step. 

The next failure is missing pinmap. The error is ERROR: Failed to parse the pinmap CSV file! 

There is a pinmap_qlf_k4n8_umc22.csv in the folder. I updated the command to: ql_symbiflow -compile -d ql-pp3 -t top -v counter_16bit.v -p counter_16bit.pcf -P pinmap_qlf_k4n8_umc22.csv

Still produces same error. I assume that I have the wrong CSV. Please advise on where to find information for PP3 pinmap or the correct CSV file.

Attached is the log output
[top.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/7206996/top.zip)
.

anthonyle-ql, 2021-09-22:
I was able to get a little further by doing the followings:

1. Specified the -P to WD30 (which is the supported package)
2. Remove all of the IO settings in the PCF file and only keep set_io clk A3

Update command is: ql_symbiflow -compile -d ql-pp3 -v counter_16bit.v -t top -P WD30 -p counter_16bit.pcf

The tool passes VPR and still fail. See attached log.

[top.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/7207112/top.zip)

anthonyle-ql, 2021-09-22:
I was able to get a little further by doing the followings:

Specified the -P to WD30 (which is the supported package)
Remove all of the IO settings in the PCF file and only keep set_io clk A3
Update command is: ql_symbiflow -compile -d ql-pp3 -v counter_16bit.v -t top -P WD30 -p counter_16bit.pcf

The tool passes VPR and still fail. See attached log
[top.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/7207121/top.zip)
.

mkurc-ant, 2021-09-22:
@anthonyle-ql Could you attach the PCF file you use?

anthonyle-ql, 2021-09-22:
Please see attached .pcf file.

If it is not too much trouble, please share your .pcf file and the command that you have used to test the design using PP3 device. This would help to reduce the communication cycle. Thanks.

[counter_16bit.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/7211547/counter_16bit.zip)

mkurc-ant, 2021-09-22:
I can see that the constraint you have is incorrect:
```
set_clk clk[0] A3
```

You should use `set_io` to constrain IO ports:
```
set_io clk[0] A3
```

I've created a PR with a fix to the IO constraint creation script: https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/542. This will fix the Python script crash but you still need to correct the PCF.

anthonyle-ql, 2021-09-22:
I am able to get a successful run. I fixed the PCF: change clock name from clk[0] to clk and set_clk to set_io. The original PCF file needs update (as you have mentioned).

I will move forward to validate the generate bit file on HW.

closed 2021-09-22 by anthonyle-ql