QuickLogic-Corp/symbiflow-arch-defs #549: can't find qlfasm command for ql-eos-s3 device

anthonyle-ql, 2021-09-24:
Executing command: ql_symbiflow -compile -d ql-eos-s3 -P PD64 -v counter_16bit.v -t top -p counter_16bit.pcf shows error:

Writing Implementation FASM: top.fasm
The entire flow of VPR took 6.36279 seconds.
FASM extra: top_fasm_extra.fasm
/home/qlblue/tools/symbiflow_daily/quicklogic-arch-defs/bin/symbiflow_generate_bitstream: line 69: qlfasm: command not found

See attached PCF file and log file

[top.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/7221962/top.zip)
[counter_16bit.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/7221966/counter_16bit.zip)

anthonyle-ql, 2021-09-27:
The same error message also applies to pp3 device. There is no bit file generated due to this error.

mkurc-ant, 2021-09-27:
@anthonyle-ql SymbiFlow depends on the `quicklogic_fasm` Python package to be installed in the system. This can be done via the following command:

```
pip3 install --user git+https://github.com/QuickLogic-Corp/quicklogic-fasm.git#egg=quicklogic_fasm
```

anthonyle-ql, 2021-09-27:
@mkurc-ant , I tried the step and it still does not work. Thanks.

closed 2021-09-28 by anthonyle-ql