QuickLogic-Corp/symbiflow-arch-defs #555: Error when PnR run on Counter_16bit design

rakeshm75, 2021-09-24:
counter_16bit design with pcf file gives the following error :

Error 1: 
Type: Placement file
File: top_constraints.place
Line: 5
Message: Invalid line 'ERROR: No GMUX location fro input CLOCK pad for net 'enable' at (32, 36, 0)' in file

PCF file:
set_io clk F3
set_io enable F1
set_io reset E4

[counter_16bit.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/7227443/counter_16bit.zip)

mkurc-ant, 2021-09-27:
@rakeshm75 I managed to reproduce the issue. There is a bug in the placement constraints generation script. I've created a PR with a fix: https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/558

mkurc-ant, 2021-11-15:
As https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/558 is merged now and a design with a counter driven by a global clock is part of the CI test for the installed toolchain this issue should be solved.