QuickLogic-Corp/symbiflow-arch-defs #557: Error when running design with FIFOs

rakeshm75, 2021-09-27:
I am getting the following error when running attached design with FIFOs. It generates the eblif file but when running the PnR it gives the error:

Error 1: 
Type: Packing
File: /home/runner/work/conda-eda/conda-eda/workdir/conda-env/conda-bld/vtr-optimized_1628838743670/work/vpr/src/pack/cluster.cpp
Line: 2376
Message: Can not find any logic block that can implement molecule.
	Atom i2c_register_read_fifo_FIFO_INST_genblk1_genblk1_genblk1_genblk1_genblk1_genblk1_genblk1_genblk1_genblk1_genblk1_RAM_0 (RAM_CE0_FE1_D0_PR0_WSA0_WSB0_VPR)

[design.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/7233311/design.zip)

mkurc-ant, 2021-09-27:
@rakeshm75 We are aware of this issue and working towards fixing it. Basically the techfile for `PP3` mentions many `RAM` pins as non-routable while they actually are routable (as in `EOS-S3`).

mkurc-ant, 2021-11-15:
This issue should be now solved via https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/613