QuickLogic-Corp/symbiflow-arch-defs #677: VQ file: bus signal are split into individual signals

rakeshm75, 2021-11-15:
In the VQ file (Verilog post-layout netlist), the bus signals are split into individual signals:

Its should be:
   output [15:0] count;

In VQ file:
    output \count[0] ,
    output \count[1] ,
    output \count[2] ,
    output \count[3] ,
    output \count[4] ,
    output \count[5] ,
    output \count[6] ,
    output \count[7] ,
    output \count[8] ,
    output \count[9] ,
    output \count[10] ,
    output \count[11] ,
    output \count[12] ,
    output \count[13] ,
    output \count[14] ,
    output \count[15] 

[counter_16bit.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/7538489/counter_16bit.zip)

mkurc-ant, 2021-11-15:
@rakeshm75 @lnsharma This is done intentionally. The problem is that for post-layout simulation with Icarus Verilog all cell ports (including top-level) must be 1-bit wide because Icarus does not support SDF annotations for wide ports. There is even an issue reported in its main repository: https://github.com/steveicarus/iverilog/issues/509

lnsharma, 2021-11-16:
> @rakeshm75 @lnsharma This is done intentionally. The problem is that for post-layout simulation with Icarus Verilog all cell ports (including top-level) must be 1-bit wide because Icarus does not support SDF annotations for wide ports. There is even an issue reported in its main repository: [steveicarus/iverilog#509](https://github.com/steveicarus/iverilog/issues/509)

@mkurc-ant : Is it feasible to generate 2 vq files - one with all cell ports of 1-bit wide and other with vector representation of cell ports? These files can  then be accordingly used in different simulators?