QuickLogic-Corp/symbiflow-arch-defs #766: Compilation error in the new output Verilog netlist file (top_post_synthesis.no_split.v file)

rakeshm75, 2022-01-09:
With the new VQ file (top_post_synthesis.no_split.v) supporting the bus signals, we get syntax error when compiling in the simulator:

# ** Error: D:/Tamar2/System_Engineering/AntMicro/QLSOFA/OpenFPGA-ArcticPro3/pre_main_physical/SymbiFlow/k4n8/postlayout_sim/counter_16bit/top_post_synthesis.no_split.v(1350): (vlog-2730) Undefined variable: 'in'.
# ** Error: (vlog-13069) D:/Tamar2/System_Engineering/AntMicro/QLSOFA/OpenFPGA-ArcticPro3/pre_main_physical/SymbiFlow/k4n8/postlayout_sim/counter_16bit/top_post_synthesis.no_split.v(1350): near "[": syntax error, unexpected '[', expecting ')'.
# ** Error: (vlog-13069) D:/Tamar2/System_Engineering/AntMicro/QLSOFA/OpenFPGA-ArcticPro3/pre_main_physical/SymbiFlow/k4n8/postlayout_sim/counter_16bit/top_post_synthesis.no_split.v(1360): near "[": syntax error, unexpected '[', expecting ')'.
# ** Error: (vlog-13069) D:/Tamar2/System_Engineering/AntMicro/QLSOFA/OpenFPGA-ArcticPro3/pre_main_physical/SymbiFlow/k4n8/postlayout_sim/counter_16bit/top_post_synthesis.no_split.v(1370): near "[": syntax error, unexpected '[', expecting ')'.

These errors occur because in the frac_lut4_arith port map  has .in[], which gives syntax error. Again either we should not split the bus signal here as well….   .in({frac_lut4_arith_count_adder_lut4_in_1_input_0_3 ,frac_lut4_arith_count_adder_lut4_in_1_input_0_2 ,  
    frac_lut4_arith_count_adder_lut4_in_1_input_0_1, frac_lut4_arith_count_adder_lut4_in_1_input_0_0 }) or have “\”in the port map .\in[3] (\frac_lut4_arith_count_adder_lut4_in_1_input_0_3 ), .\in[2]…

[counter_16bit.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/7835229/counter_16bit.zip)

tpagarani, 2022-01-10:
@mkurc-ant, could you please look into this on priority?

mkurc-ant, 2022-01-10:
@tpagarani Sure, I can work on solving that. @rakeshm75 Could you point me to the git revision you are using and if possible to the design that exposes the issue too?

rakeshm75, 2022-01-10:
@mkurc-ant, I am using the latest Symbiflow version. I have attached the counter_16bit design here which can reproduce the issue. 

[counter_16bit.zip](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/files/7838227/counter_16bit.zip)

mkurc-ant, 2022-01-10:
@rakeshm75 Thanks.

lpawelcz, 2022-01-14:
https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/777 is now merged. It should fix your problems @rakeshm75. Please check your design with updated symbiflow.